SequoiaSan/FlipperZero-WiFi-Scanner_Module #3: Sort scan list by RSSI

SequoiaSan, 2022-07-15:
I didn't done this, because the list itself is managed by ESP API and there is no option to sort it. 
Need to investigate.

ramsesramosgzz, 2022-07-20:
when i click on an SSID it doesnt go to monitor mode it just goes back to another SSID am i missing something?

SequoiaSan, 2022-07-20:
@ramsesramosgzz 
Thanks for participation!

1. I assume that you're pressing the centre button in the navigation ring (aka OK button) when one of the SSIDs in focus on the scan page? Right?
2. Did you compile code by yourself or used [this release](https://github.com/SequoiaSan/FlipperZero-WiFi-Scanner_Module-ESP8266/releases/tag/v.0.1)?
3. Which scheme did you use? With a voltage regulator or direct 5V? (it might be the case that some chips don't work properly on FlipperZero's 5V. It's still under investigation)
4. Which ESP8266 do you use? Mini, Mini Pro or something else?

I'll try to provide you with debug steps that will nail down the issue.

SequoiaSan, 2022-07-21:
[Added scan sort by RSSI](https://github.com/SequoiaSan/FlipperZero-WiFi-Scanner_Module/commit/ac9b5995f1616c5c493ccb035d8b47fbe9a9817e)

closed 2022-07-21 by SequoiaSan