SequoiaSan/FlipperZero-WiFi-Scanner_Module #16: Can't flash esp32s2

Brightyfalls, 2022-09-16:
Hi!  FlipperZero-WiFi-Scanner_Module.ino.esp32s2.bin won't flash properly on flipper dev board v1. I've followed the instructions provided [here](https://github.com/SequoiaSan/Guide-How-To-Upload-bin-to-ESP8266-ESP32) but every time I flash, I get the error attached below. Thank you in advance for any help! 

[image]

SequoiaSan, 2022-10-02:
Hi Brightyfalls!

Have you checked your cable? The error might be related to a broken cable. Please, switch the cable to another one and try again. And please report to this thread if this fixed the issue.

Thanks!

Brightyfalls, 2022-12-19:
Hi again! 

Sorry for the late reply. I have tested it on a separate machine with a different cable and it worked like a charm. Thank you for the suggestion!

[image]