SequoiaSan/FlipperZero-WiFi-Scanner_Module #17: Doc is not up to date

pierrbt, 2022-10-01:
https://github.com/SequoiaSan/FlipperZero-WiFi-Scanner_Module#how-to-add-new-application-on-flipperzero

I tried to add the application to Firmware but [applications/meta/application.fam](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/applications/meta/application.fam) do not exists anymore.

SequoiaSan, 2022-10-02:
Hi pierrbt!

Thank you for pointing this out. I haven't had enough time last days, because of the war here in Ukraine, but I'll post a new release of Scanner as a standalone app and update the Readme.

pierrbt, 2022-10-02:
You mean a .fap file ?
And ok i understand, good Luck for the War 🇺🇦

SequoiaSan, 2022-10-02:
Yes, .fap file. Haven't had a chance to look how now they propose to compile apps in *.fap
When this happen I'll post new release + update docs accordingly.

Thanks! ❤️